NEVSTOP-LAB/Communicable-State-Machine #106: JKISM++与JKISM和JKISMO之间的对比

huipeng8, 2023-08-10:
如题，请问有他们之间的对比说明吗？谢谢

nevstop, 2023-08-10:
@huipeng8 这个 repo 最近正在开发中，所以还没有完善 wiki 和全部文档，这个问题后面会更新到文档中, 计划对比的是 DQMH/SMO

简单的说：

- JKISM 是一个单功能开发的模板，用于开发某个界面或功能，不是程序框架；
- JKISMO 是 JKI 基于 JKISM + OOP 设计的程序框架，很强大，但是由于内部太复杂，很多新手不容易上手，对于开发者的要求很高；
- JKISM++ 是我根据 JKISM 拓展的编程框架，依然延续 JKISM 没有类的设计，通过添加一些新的字符串规则，实现不同 JKISM 之间的消息交互，熟悉 JKISM 的开发者应该很容易上手。可以看下 wiki 的一些内容，虽然还是施工中;范例也有很多了，可以安装后试试看。
语法参考：[build-state-string-with-argumentsvi](https://github.com/NEVSTOP-LAB/JKI-State-Machine-Plus-Plus/blob/main/src/help/NEVSTOP/JKI%20State%20Machine%2B%2B/VI%20Description(CN).md#build-state-string-with-argumentsvi)

nevstop, 2023-08-27:
Create stub in wiki.

closed 2023-08-27 by nevstop

nevstop, 2023-10-08:
@huipeng8 通过和 JKI的沟通，Library 名字修改为 Communicable State Machine(CSM), 也已经发布到 vipm.io. 请知悉。 欢迎在项目中使用并提供反馈哈。