NEVSTOP-LAB/Communicable-State-Machine #359: 降低模板 头部空间空间优化建议

casey502, 2025-01-05:
模板可否再进行优化一下空间，具体如下图所示。
[image]

优化后的建议如下图，降低 template 头部空间：
[image]

这样可以减少上面空间的浪费，对整个模板空间有用的利用率更高，尽量的减少上方的空间。
因为一般编写具体实现的代码，都是从左到右，从下到上的方式比较利于代码的整洁。

nevstop, 2025-01-05:
已经修改，2025一月发布的版本中将包含

nevstop, 2025-01-06:
[image]

nevstop, 2025-01-06:
[image]

nevstop, 2025-01-06:
Convert to discussion for keeping this thread easy to find by other users.